adafruit/Adafruit_SHT31 #2: Adafruit_SHT31::begin always returns true

anecdata, 2016-01-09:
Commented out:
//return (readStatus() == 0x40);

Testing for presence is useful during setup.

anecdata, 2016-01-13:
It appears that the status after reset should be 0x8010: bit 15 indicates an alert status is pending (including various levels of reset), and bit 4 indicates reset detected. Some discussion of SHT31 presence testing at (http://forums.adafruit.com/viewtopic.php?f=22&t=87407)

anecdata, 2016-02-09:
Seeing an 0x8050 status in the wild now, which is unusual since a reserved bit is now set.

anecdata, 2016-02-09:
Suggestion to use serial number as a startup success test:

[https://forums.adafruit.com/viewtopic.php?f=22&t=87407#p450345](https://forums.adafruit.com/viewtopic.php?f=22&t=87407#p450345)

Which seems consistent with Sensiron's sample code:

[https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/Sample_Codes_Software/Humidity_and_Temperature_Sensors/Sensirion_Humidity_and_Temperature_Sensors_SHT3x_Sample_Code_V2.pdf](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/Sample_Codes_Software/Humidity_and_Temperature_Sensors/Sensirion_Humidity_and_Temperature_Sensors_SHT3x_Sample_Code_V2.pdf)

acekrystal, 2016-12-19:
I changed the original "sht31.begin" to return correct true or false using the already existing "sht31.readStatus" function.

I basically tested what code "sht31.readStatus" returns in decimals when sensor is not connected, and used that for a simple if statement.

in Adafruit_SHT31.cpp new "begin" function changes:
```
boolean Adafruit_SHT31::begin(uint8_t i2caddr) {
  Wire.begin();
  _i2caddr = i2caddr;
  reset();
  //return (readStatus() == 0x40);
+  if (readStatus() != 65535){ // None connected sensors seem to return decimal 65535.
+	  return true;
+  }else{
+	  return false;
+  }
-  return true;

}
```

hoffmannjan, 2019-07-18:
Added in #19, thanks @acekrystal ! 👍

closed 2019-07-18 by hoffmannjan