adafruit/Adafruit_SHT31 #10: ESP8266 communication issues

drtrigon, 2017-10-20:
- Arduino board:  ESP8266 (ESP-01 Modul)

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.3 (esp8266 lib version 2.3.0)

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): I combined your SHT31 example with a REST example to get a REST API output for the SHT31, please confer the code. No matter what I try (I2C pullups, etc.) the SHT31 example works on an Arduino DUE but the ESP8266 cannot communicate with it. The I2C communication seems to work but returned values are always nan. readStatus always returns 65535.

```
/*
  This a simple example of the aREST Library for the ESP8266 WiFi chip.
  See the README file for more details.

  (Written in 2015 by Marco Schwartz under a GPL license.)

// SHT31
// This code is designed to work with the SHT31_I2CS I2C Mini Module available from ControlEverything.com.
*/
/*************************************************** 
 *    TODO:
 *      - sht31: not working...
 *               
 *      - test LED output setting ".../led/o"
 *      - LED switch OFF ".../led/1" does not work! (ON works)
 *        command string does not contain the value, try e.g. ".../led?1"
 *        same with ".../digital/1/w/0",  ".../digital/1/w/1" works, test  ".../digital/1/r"
 *      - still need "set fuse settings (clock, etc.)" ? (Tools > Burn Bootloader)
 *    
 *    Programm ESP8266 using Arduino Due Board as 3.3V USB2serial converter:
 *    - Links
 *      - Hardware: https://www.itead.cc/wiki/ESP8266_Serial_WIFI_Module#Hardware
 *      - Arduino IDE mit dem ESP8266: https://arduino-hannover.de/2015/04/08/arduino-ide-mit-dem-esp8266/
 *      - ESP8266 with SHT31(-D) WiFi WebServer: https://github.com/ControlEverythingCommunity/SHT31/blob/master/ESP8266/SHT31.ino
 *                                               https://www.controleverything.com/content/Humidity?sku=SHT31_I2CS#tabs-0-product_tabset-2
 *      - ESP8266 I2C example: http://pdacontrolen.com/esp8266-display-oled-i2c-client-irc/ (SCL-GPIO0, SDA-GPIO2)
 *      - https://learn.adafruit.com/esp8266-temperature-slash-humidity-webserver/uploading-to-the-esp8266
 *    - use Arduino IDE >=1.8.3
 *    - install ESP8266 option to IDE
 *      - File > Preferences > Additional Boards Manager URLs: http://arduino.esp8266.com/stable/package_esp8266com_index.json
 *      - Tools > Board: ??? > Boards Manager...
 *        search for: "esp8266" and install (esp8266 2.3.0)
 *    - connect ESP8266 to an Arduino Due by wiring:
 *        Arduino Due Pin   ESP8266 Pin
 *                    GND   1 GND
 *                 TX0->1   2 UTXD          --------------------
 *                          3 GPIO2         |2 1        x--=== |
 *               3.3V VCC   4 CH_PD         |4 3           === |
 *                    GND   5 GPIO0         |6 5  ESP8266  === |
 *                          6 RST           |8 7  board      | |
 *                 RX0<-0   7 URXD          --------------------
 *               3.3V VCC   8 VCC
 *        Arduino Due RESET pin needs to be connected to GND (in order to disable the chip/sketch)
 *        (ESP8266 RST kann offen bleiben oder PullUp)
 *    - select ESP8266:
 *      - Board: "Generic ESP8266 Module"
 *      - Flash Mode: "DIO"
 *      - Flash Frequency: "40MHz"
 *      - CPU Frequency: "80MHz"
 *      - Flash Size: "512K (64K SPIFFS)"
 *      - Debug port: "Disabled"
 *      - Debug Level: "None"
 *      - Reset Method: "ck"
 *      - Upload Speed: "115200"
 *      - Port: ???
 *    - select Programmer:
 *      - Tools > Programmer: "AVRISP mkII"
 *        (this is the default for Arduinos)
 *    - upload this code
 *        (if it does not work dis- and re-connect the Due/ESP8266 or try the RST pin)
 *    
 *                      ESP8266 Pin Layout
[... 199 lines cut ...]
    data[3] = Wire.read();
    data[4] = Wire.read();
    data[5] = Wire.read();
  }
  // Convert the data
  int temp = (data[0] * 256) + data[1];
  //cTemp = -45.0 + (175.0 * temp / 65535.0);
  temperature = -45.0 + (175.0 * temp / 65535.0);
  //fTemp = (cTemp * 1.8) + 32.0;
  humidity = (100.0 * ((data[3] * 256.0) + data[4])) / 65535.0;

  // Output data to serial monitor
  Serial.print("Temperature in Celsius :");
  //Serial.print(cTemp);
  Serial.print(temperature);
  Serial.println(" C");
  //Serial.print("Temperature in Fahrenheit :");
  //Serial.print(fTemp);
  //Serial.println(" F");
  Serial.print("Relative Humidity :");
  Serial.print(humidity);
  Serial.println(" %RH");
  delay(500);
}*/
```

drtrigon, 2017-10-21:
Could that be related to issues with repeated start?

hoffmannjan, 2019-07-18:
tested and It works - please do an update and reopen if still an issue. 👋

closed 2019-07-18 by hoffmannjan