adafruit/Adafruit_SHT31 #38: Add function for reading serial

Ozelod, 2023-07-19:
Is it possible to add this function?
[https://sensirion.github.io/python-i2c-sht/0.2.0/_modules/sensirion_i2c_sht/sht3x/commands.html](https://sensirion.github.io/python-i2c-sht/0.2.0/_modules/sensirion_i2c_sht/sht3x/commands.html#Sht3xI2cCmdReadSerial.__init__)

Adafruit_SHT31.h
```
#define SHT31_READSERIAL 0x3780
...
class Adafruit_SHT31 {
public:
  Adafruit_SHT31(TwoWire *theWire = &Wire);
  ~Adafruit_SHT31();

  ...
  uint32_t readSerial(void);
  ....
```

Adafruit_SHT31.cpp
```
uint32_t Adafruit_SHT31::readSerial(void) {
  writeCommand(SHT31_READSERIAL);
  
  delay(10);
  
  uint8_t reply[6];
  if (!i2c_dev->read(reply, sizeof(reply))) {
    return false;
  }

  if ((crc8(reply, 2) != reply[2]) || (crc8(reply + 3, 2) != reply[5])) {
    return false;
  }

  uint32_t serial = 0;
  serial = reply[0];
  serial <<= 8;
  serial |= reply[1];
  serial <<= 8;
  serial |= reply[3];
  serial <<= 8;
  serial |= reply[4];

  return serial;
}
````

ladyada, 2023-07-19:
please submit a PR!

Ozelod, 2023-07-21:
thanks, i did it.

closed 2023-07-21 by Ozelod